4712/BLHeliSuite #1: ATmega2560 1-Wire Interface Bootloader Error

cfunseth, 2015-07-30:
When attempting to create a 1-wire interface with an Arduino MEGA 2560, I run into the following error when attempting to flash the bootloader onto the board:

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.03s

avrdude: Device signature = 0x1e9801
avrdude: NOTE: FLASH memory has been specified, an erase cycle will be performed
To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: stk500v2_command(): command failed

avrdude done. Thank you.

I'm plugged in directly over USB from the computer to the 2560 if that matters.  I've found a similar issue reported here when attempting to erase the 2560 using avrdude: https://github.com/arduino/Arduino/issues/1507

I can post screenshots of the error tonight if need be but wanted to see if there was an easy way to add the "-D" parameter to the avrdude command being sent from BLHeliSuite.

4712, 2015-07-31:
Try this exe:

http://www.mediafire.com/download/hi894vvoxr7l12a/BLHeliSuite.7z
Next update will add -D option for all Arduino boards.

cfunseth, 2015-07-31:
Thank you, this worked perfectly!

4712, 2015-08-10:
BLHeliSuite14.0.0.3 now is released and uses -D option for all Arduino boards.

closed 2015-08-10 by 4712