4712/BLHeliSuite #2: Cross-Platform Support

tracernz, 2015-08-09:
I have no idea what this is written in/what libraries it uses but is there any possibility of *nix support? I'd really like to be able to use my laptop (OS X) or Linux.

4712, 2015-08-09:
BLHeliSuite is written with Delphi 2006 for Windows. An upgrade to Delphi XE for Multiplatform is beyond my capabilities....

tracernz, 2015-08-09:
Ah, that's unfortunate. Certainly understandable that you don't wish to upgrade. :wink:

closed 2015-08-09 by tracernz

Bauerpauer, 2015-09-08:
@4712, is it the time you'd need to invest, or the money needed to purchase Delphi XE?  I'd bet between rcgroups.com and /r/multicopter funds could be donated to the cause.  Any chance you could open-source the Delphi code so that someone could at least take a crack at porting it to a cross-platform codebase?

SleepyBrett, 2015-10-23:
Or just release the source so someone with delphi xe can cross compile it or someone else can rewrite it using the original as a guide.

4712, 2015-10-24:
A release of the source is not planned atm.
I'll happily try to reply all detailed technical questions (as far as I can and spare time is leftover).

4712, 2015-10-28:
BLHeliSuite14.2.0.0 released.
Added support for basic functionality under Linux/Wine (might be also OSX/Wine ). Please see “..\Manuals\BLHeliSuiteWine.pdf” or https://github.com/4712/BLHeliSuite/blob/master/Manuals/BLHeliSuiteWine.pdf

tracernz, 2015-10-28:
Thanks, @4712. I'll test this out when I get a chance. :+1:

JasonDorie, 2016-04-07:
How possible would it be for someone to alter BLHeli ESC settings from a flight controller?  I ask because I'm writing flight controller firmware, and it would be nice for us to be able to apply our default ESC settings directly from the controller instead of having to use an external device.  And I personally would like to be able to disable the "beacon" feature on the bench (when the controller is connected to a host PC) but then enable it when flying normally.

I'm totally willing and able to do the implementation myself, but I can't find much in the way of protocol details for how I'd go about it, or whether it's even possible.  It would be a bonus if I was able to do a full firmware update through our flight controller (it's not Arduino compatible), but I assume that's harder.  Any guidance you're willing to offer would be much appreciated.

4712, 2016-04-10:
All is possible, you might have a look at the implantation of the 4way-if in CF/BF and raceflight. https://github.com/4712/cleanflight/tree/blheli4wayif
I would not suggest to do it the way you described. The risk that something fails is always given and you would not get a failure message.