4712/BLHeliSuite #4: Allow exclusion of some parameters from multi-ESC sync

joshuabardwell, 2015-10-05:
It would be nice to be able to exclude PPM min, PPM max, and motor direction from the multi-ESC sync determination, as desired. For example, I would like to reverse all my motor directions, without over-writing my current PPM min and max values. I would like my ESCs to show as in sync even though they have different PPM min and max, or motor direction.

4712, 2015-10-05:
motor direction is already excluded...

4712, 2015-10-11:
For the next release:
"MultipleESC function now allows optional exclusion of “Motor Direction” and “Min/Center/Max
Throttle” from Master-Slave synchronization. Check or uncheck sync in menu “Options... Sync….)."

joshuabardwell, 2015-10-11:
Thanks much.

closed 2015-10-11 by joshuabardwell