4712/BLHeliSuite #9: 4-way Silabs suite hangs

hsuq510, 2016-05-31:
Using the 4-way interface to read a Silabs ESC (LB 20Pro) the Blheli suite hangs. The 14.4 still works fine. I tried to reprogram the NANO 328 with the interface inside 14.5 but issue still occurs.

4712, 2016-05-31:
What file did you flash to the NANO 328?
Please post log.tlg.

hsuq510, 2016-05-31:
I have tested all 4wArduino_Nano__16* and also the Pro mini with 328. It connects and square appear, but as soon I read setup, blheli suite hangs. Instead im able to read setup with the same Nano 328 within the 14.4 suite and the with file Arduino_Nano__16_Multi using D3 and GND pin.

Since the blheli is not responding I cant save the log.tlg. I can attach the screenshot.

[image]

hsuq510, 2016-05-31:
Ok, I was able to use 1-wire instead. But with 14.4 suite the 4-way interface was working and no need to reprogram Nano to use USB/Com.

4712, 2016-06-01:
@hsuq510 For the LB 20A Pro, you can use the 4wArduino_m328P_16_MULTIv14400.hex file also with the newest BLHeliSuite. Did you try this?

4712, 2016-06-01:
@hsuq510 Ok found the issue (sync objects in XP and Wine), it will be fixed in the next release. Thanks!

4712, 2016-06-08:
Hopefully fixed in rev 16.1.14.6.0.0

closed 2016-06-09 by 4712