4712/BLHeliSuite #11: how can i add a type of esc to blhelisuite as a default firmware

zhangxinwei0212, 2016-06-23:
hi ,can you tell me how to add a new type of est to blhelisuite as a default firmware? please! thank you very muck! Best wishes

4712, 2016-06-24:
@zhangxinwei0212 does the ESC use BLHeli (Atmel or SiLabs) or BLHeli_S?

zhangxinwei0212, 2016-06-28:
The esc use BLHeli of SiLabs  now i want added my esc hex to BLHeliSuite as a default firmware.

4712, 2016-06-28:
@zhangxinwei0212 I've found your BLHeliSuite fork with the added ESC.
If your ZTR20A is a perfect match with the already existing LittleBee 20A code, we’ll recommend to use that code. If you have more questions please feel free to contact me at RCGroups user 4712.

zhangxinwei0212, 2016-07-06:
ok, thank you very much  have a good time !!!

closed 2016-07-06 by 4712