4712/BLHeliSuite #12: 1-Wire BLHeli_S kill Seriously Pro Evo

Sijjim, 2016-06-27:
Hi
I build a new Quad with Seriously Pro Evo and Emax Lightning 35A (BLHeli_S) and want to change Settings in these ESC.
Previously i did it with "older" Hardware and all done perfect but with this combo (SP Evo & BLHeli_S) the Flightcontroller was destroyed. 
Flightcontroller was flashed with actual Cleanflight (1.13.0), connected all, open BLHeliSuite, choose Interface "E" - SILABS BLHELI BOOTLOADER (Cleanflight), choose Com-Port and push "Connect". Little Window open with Massage like "Com Port closed, Com Port not found..." i dont read exactly, at this point i tought a misstake at my side.
But since here no computer know anything is connected, if i put this Evo-Board in. I tried to boot the Evo in Bootloader but nothing, its really dead. Later i do same procedure (all the same but the Flightcontroller) with an normal Naze32, all done without problems.

I guess, the problem is, because the USB-Bridge from Controller on Evo-Board needs also other Windows-Driver (Zadig | STM ) to connect with Cleanflight-Configurator and BLHeli dont know that.

4712, 2016-06-27:
I'm sorry that you have such difficulties.
BLHeliSuite can't kill any FC board while trying to connect over VCP.
Have a look at the EVO manual, how to connect in bootloader mode.
Or ask the seller, what to do in your situation.

Sijjim, 2016-06-27:
I dont want a new FC from you or a repair Service ;) I know its open Source and use in own risk....
EVO manual say, close 2 Pins to enter bootloader mode, but it dosn't.
Thats not why i wrote here. 
I dont understand, where the misstake is. I do this procedure much times in past with much different Hardware, thats all not new for me. What I dont understand: click on "Connect" in BLHeliSuite and FC is dead. Any hint, where i look for fault?

Thanks

4712, 2016-06-27:
Any LED on or flashing?

Sijjim, 2016-06-27:
On EVO? On Power up (USB or direclty) only green LED on, without any change, blink or any others. Other serials also dont answer.

4712, 2016-06-27:
Does the device manager show any unknown USB device?

Sijjim, 2016-06-27:
no, nothing, windows dont register anything is connected. Tested in Linux, same problem. Tested in booth modes, same problem.

4712, 2016-06-27:
Maybe the vreg is dead...

Sijjim, 2016-06-27:
You are 100% sure, the error can't triggered from BLHeliSuite? What can I do to protect for the same error again?

4712, 2016-06-27:
CFConfigurator uses MSP commands and BLHeliSuite uses them. If you got an error message on trying to connect, there were no commands executed at all...

Sijjim, 2016-06-27:
this happened:
I flashed EVO to cleanflight 1.13.0, setup completly Board (Modes, Receiver...) and closed Cleanflight Configurator. Then I opened BLHeliSuite and only want to change Motor directory but at the moment I pushed the "Connect" button in BLHeliSuite, the Windows sound "USB Disconnected" plays and Alert-Window with Message "Com Port closed, Com Port not found..." open. Since this point the FC do the things what I have described above.

Understandably I dont trust the BLHeliSuite now and think there is the error.

Sijjim, 2016-06-27:
Maybe anybody else has been tested exactly this combo?
Yes, i will move to RCG.

Sijjim, 2016-07-26:
sorry for late reply.
I was brave and tried the same combo again with success, all done. But the first SPro Evo is irretrievably destroyed and I still do not know why...

closed 2016-07-26 by Sijjim