4712/BLHeliSuite #13: Are you willing to support sending 4WIF data over TCP / named pipe?

mlyle, 2016-08-21:
Hi, I'm a developer for dRonin.  We're looking at putting together BlHeli pass-through capabilities, and we could do it similar to CF using pass-through VCP.  But what we'd really like to do is mediate the communication through our telemetry protocol and ground control suite--- so we can do the update over telemetry radios or network connections.  Ability to tunnel serial-like protocols through our telemetry protocol could be useful for other things, too (GPS RTK, control of other devices in flight, etc)

There's things like this http://com0com.cvs.sourceforge.net/viewvc/com0com/com2tcp/ but they'd be a pain for our users.

Would BlHeliSuite be willing to send the programming data over a network or network-like connection in addition to sending it directly to serial ports?  Is this easy to abstract and achieve?

references: d-ronin/dRonin#104

4712, 2016-08-21:
@mlyle. I will have a look at it the next days. Depends on the amount of time needed...
What do I need to establish a connection for testing?

mlyle, 2016-08-21:
@4712 we don't have our side implemented.  There are TCP-to-serial relay drivers that could be used to talk to an existing TCP dongle, perhaps, but they sound like a pain to get going... I would be willing to put an environment like that together and test builds.

Alternatively, if you have another program that can make a listening socket-- like netcat-- just making sure that it can form the connection, send the first characters in the protocol, and properly block for a response would validate a lot of the functionality.   Something like this would work: https://sourceforge.net/projects/sockettest/ 

I hope it's not too much work and I'm glad to pitch in any way I can.  Thanks for your consideration!