4712/BLHeliSuite #16: Feature suggestion

sa-mendez, 2017-03-04:
I would like to suggest the following addition to BLHeliSuite UI:

- Add **MCU Class** information row to the ESC Overview table to show the MCU Class of each ESC.

Motivation:

It would be nice to be able to verify that an ESC is actually using the MCU you expect.  
In my case, I bought a couple of LittleBee 20a Pro ESCs which are supposed to have an F39x MCU.  It turns out that one of them actually had an F33x MCU.  
I did find this out when I attempted to flash it with the Multishot version of the firmware, and BLHeliSuite gave me an appropriate error indicating a mismatch between the MCU required by the firmware and the one that was in the actual ESC.

The MCU information appears to be readable by BLHeliSuite, so I think this would be straight forward to just add to the table in the ESC Overview tab.

Thank you,

Sergio

P.S.

I am willing to assist with this, but I cannot seem to find the BLHeliSuite source code in GitHub (just the .exe) in this repository.

4712, 2017-03-31:
@sa-mendez: It is not that easy, because using the bootloader, there is now way to read the MCU directly. The bootloader is compiled with the MCU data. So, if the wrong bootloader is flashed, BLHeliSuite "sees" the wrong MCU.

sa-mendez, 2017-04-01:
@4712:  I see what you mean.  Makes sense.  The fact that attempting to flash traps the error is good enough.

closed 2017-04-01 by sa-mendez