4712/BLHeliSuite #21: Cant open COM port

trashmaster9000, 2017-07-23:
Hi,

i have a problem with BLHelisuite 16.6.14.9.0.2:

it doesnt recognize COM ports.
i cant select the Interface.

i have the arduino COM port in the device manager.

do i need to install runtime libraries?

4712, 2017-07-23:
If it is shown as a fully functional VCP device in the device manager, it will be listed in BLHeliSuite as well.
What arduino board is it, what USB/serial chip is used and how does it show up exactly?

trashmaster9000, 2017-07-23:
its a cheap Arduino nano clone with the CH340 chip.
it worked last time it updated my ESCs (i think at version 14.6, but not shure).

[image]
[image]
[image]

trashmaster9000, 2017-07-23:
i tried my USB2Serial converter with a Prolific PL2303 chip.

blhelisuite recognizes this chip but not the CH340. it worked a few months ago with this exact Arduino nano...

4712, 2017-07-23:
Could you please try 
[DumpPorts.zip](https://github.com/4712/BLHeliSuite/files/1168179/DumpPorts.zip)

trashmaster9000, 2017-07-23:
`\Device\Serial????????    REG_SZ    COM5
\Device\ProlificSerial0    REG_SZ    COM6`

Arduino nano (COM5) and USB2Serial adaptor (COM6) connected.

is this a windows 10 bug?
since the last time i used blhelisuite i got the "Creators update"...

4712, 2017-07-23:
May be, I have tested with win8.1 and 7.    
with win8.1, I get:
\Device\Serial0    REG_SZ    COM1   
\Device\Serial2    REG_SZ    COM12
Com12 is my CH340.
Can you check this BLHeliSuite.exe with enabled Log?
Look for something like this:
[image]

 
[BLHeliSuite.zip](https://github.com/4712/BLHeliSuite/files/1168289/BLHeliSuite.zip)

4712, 2017-07-23:
BTW: Meanwhile I assume, that the CH340 driver uses Chinese signs, which cannot be handled by BLHeliSuite, because it is not Unicode capable. Maybe you try to reinstall or use another driver.

4712, 2017-07-24:
I guess, I found a fix, please check the exe file:
[BLHeliSuite.zip](https://github.com/4712/BLHeliSuite/files/1169950/BLHeliSuite.zip)

trashmaster9000, 2017-07-24:
ok, it works now with the new exe.

4712, 2017-07-24:
@trashmaster9000 Thanks for testing, BLHeliSuite 16.6.14.9.0.3 with the fix is [uploading ](https://www.mediafire.com/folder/dx6kfaasyo24l/BLHeliSuite)

closed 2017-07-24 by 4712