4712/BLHeliSuite #22: Creating a custom stm32 4way interface

bikemike, 2017-08-23:
I'd like to create a 4w-if like the arduino but on a stm32 platform connected to a usb-ttl. Is there some reference code I can look at to see how to proceed? It looks like the arduino version of the 4way is not open source?

4712, 2017-08-24:
I would suggest to look at betaflght sources
https://github.com/betaflight/betaflight/blob/master/src/main/io/serial_4way.c
https://github.com/betaflight/betaflight/blob/master/src/main/io/serial_4way.h
https://github.com/betaflight/betaflight/blob/master/src/main/io/serial_4way_avrootloader.c
https://github.com/betaflight/betaflight/blob/master/src/main/io/serial_4way_avrootloader.h
https://github.com/betaflight/betaflight/blob/master/src/main/io/serial_4way_impl.h
https://github.com/betaflight/betaflight/blob/master/src/main/io/serial_4way_stk500v2.c
https://github.com/betaflight/betaflight/blob/master/src/main/io/serial_4way_stk500v2.h
is all what is needed...

bikemike, 2017-08-24:
Great. Thank you.

If I were to implement this protocol, and connect using a usb-ttl would I select "SILABS BLHeliBootloader (USB/Com)" or "SILABS BLHeliBootloader (4way-if)"? And are there specific baud rate requirements?
 Would I need to use a usart or could I bitbang the serial on different pins?

4712, 2017-08-25:
You would need ""SILABS BLHeliBootloader (4way-if)", the serial port setting for the ESC is always 19200,8,n,1. You could use usart or soft serial.

bikemike, 2017-08-25:
Thank you, I will see what I can do with all this.

closed 2017-08-25 by bikemike