4712/BLHeliSuite #27: Align RCE-BL35P has no access via 1-wire

Wesie, 2019-06-22:
Progger:Arduino Nano 4way-if with recommended flashfile
BLHeliSuite : latest version
ESC Align RCE-BL35P

After soldering the 3 wire in the ESC, select the Arduino as C2, connecting the ESC on a battery and flash the BLHeli Software (the Suite is version 16.x, the Flashfirmware (Main) is version 14.x, I have an access via the 4way-if Adapter directly on the processor to the settings and also the possibility to reflash the FW.
If I try to use a SiLab USB/Com adapter, which is working with  Racerstar ESC's without any problems, I didn't receive any response from the Align ESC. The middle Red wire, I have disconnected from the Servo plug, because the ESC contains an internal BEC. I've selected also the change Com Port.
A compare between the flash Main FW and the hex file let the message pops up, that there is 1 different in the bootloader  and 1 different in the settings.
Also a reflash of the FW does not solved the different. Maybe it's the bit which looked/unlooked the BL?

Is there a change, that the ESC could be programmed wia the Signal wire so like other ESC's, too?

4712, 2019-06-22:
When the ESC responds normally to input signals, it is usually a matter of pairing with the appropriate USB adapter. The relevant difference is in the value of the pullup resistor that should match the ESC input circuit.

Which adapter do you use exactly?
How does the ESC input circuit build (which resistors)?

Wesie, 2019-06-22:
Which kind of resistor? If I look into the PDF where a pictures available of the Align RCE-BL35P, I could following any wiring on the PCB from the signal.
I've test with an USB-Linker from www.szfvt.com with a SilabsChip F326. There 2 resistors with 01B on top. One goes from the Chip over the resitir to the signal-output. The other looks like, that he is used to put the power to zero(?) Pulldown....... This one is working with the Racerstar BLHeli ESC's
Afterward I have locked into the pdf (build adapters) and recognise, that I can use the arduino, which I have programmed as SiLabs C2 ("B") can also use as singel wire. After give an attention, that the right cable is connected to the signal, I receive also here following message: Serial Interface "4wANm328P16" connected successfully. Connection to ESC#1 failed. Please check ......"
An add. test was with an Afro-USB-Linker. ( I didn't expect, that this one should run, but also her no connection was possible via the servo-cable (1-wire).

Wesie, 2019-06-22:
During compare I hae different on Adress 0x1A10 Offset 01.
In the Main-hex is the value 8F, in the ESC is the value B4.

Wesie, 2019-06-22:
About the USB_Linker from szfvt : This one is written in the description of some ESC's:

> ll components Littlebee ESC uses are imported from overseas and especailly the Mosfet are best quality in R/C model industry. The firmware is pre-flashed with Blheli or Blheli_S, our company has USB linker programmer to connect with Blhelisuite too.

4712, 2019-06-22:
@Wesie 0x1A11 is parameter value for Governor Target RPM, so, nothing relevant here.
Afro-USB-Linker is only useful for Atmel based ESC.
4wANm328P16 using interface "D" needs to be connected with D3 to signal wire and GND.
If you have one, perhaps look with an oscilloscope on the line, what is going on.

Wesie, 2019-06-22:
D3?
Let me make a deeper look on that tomorrow. I must remove the scrink to add a cable.... 
Tomorrow I'll response the result. 
So as mention, about the Afro Lunker... I was sure, that it doesn't work... but only for test all possibilities ;)

Wesie, 2019-06-23:
I've check it once again : I have used the right wire from pin D3. No response from the ESC.
The same pin is used for 3 wire programming as clock (Red wire)

4712, 2019-06-23:
@WesieI 'm afraid, I can't help furthermore on this, I have no sample to check the signal. If you do not find a way to connect via bootloader, one option would be to add a 3 pin adapter for accessing via C2 Interface.

Wesie, 2019-06-23:
The 3 wire are still connected and the isolation is done with fluid plastic.....
What is the exact procedure over the Signal-cable, that the ESC could be programmed? Is this a spezial PWM-series? Is there a spefic range, where the ESC expect this one? What are the min. start-requirements, that the ESC goes into the programming mode?

Wesie, 2019-06-23:
With a small Oszi, I can only see, that the D3 has up from the "first connect"  all time high-level (Tested with the Arduino). It doesn't matter if I press disconnect and connect again. Interface-Type = "D", hexfile is 4wArduino_Nano__16_PD3PD2v20002.hex 
Only for sure: The 1-wire feature should work also with the "MAIN" Firmware, isn't it? It's not only in the hex for "Multi" ?

4712, 2019-06-23:
Bootloader connects with rs232 19200 bps 8n1.
Arduino pin D3 should be high connected or disconnected, but on "Read Setup" it should send rs232 signals for a while until it times out. 
Bootloader works with all modes (Main-Multi-Tail).

Wesie, 2019-06-24:
OK, the Oszi showed by 20ms, that the Arduino send the some cmd's on the Pin D3 (with and without ESC)
Could it be possible, that the ESC must be see first of all a 0 Throotle Signal? That the ESC is waiting before initialize and response to the BLHeli-Suite?
 It doesn't matter, if I connect only the Lipo on the ESC (open PWM Signal wire) or if I connect the ESC on the Arduino. 
In both cases if I connect the Lipo, the ESC played only the "Power UP" Sequence "_  -  *"

4712, 2019-06-24:
@Wesie OK, then find the path of the signal wire to CPU pin and check if there is a serial, pull up or pull down resistor. You could check the signal at the MCU input pin with the osci and see what happens here finally. 

You could also to try to add a pull D3 to Arduino 3.3V or 5V with a value lets say 2.7K  (1K- 4.7K), and see if this helps

Wesie, 2019-06-25:
Which pin should it be on the MCU? Uncomfortable I didn't found a circuit diagram.
A Pull-Up resistor (2K)  D3 agains 5V didn't help.

4712, 2019-06-25:
https://www.silabs.com/documents/public/data-sheets/C8051F33x.pdf pg37
Should be P0.5 = pin 16.

Wesie, 2019-06-26:
The signal come in and goes over a resistor (1) 2,2K (SMD-Code 222). From
the resistor (1) the Signal goes one time to the MCU P0.4 and P0.5.
Additionally from the resistor (1) the signal goes parallel to a capacity
and another resistor (2) 20K (SMD-Code 203). The capacity and the resistor
(2) are connected to ground.
You can see the 2 resistor and capacity in the PDF on the picture with the
marked points for programming. The big red + wire and direct on the right
the 3 components.
The Oszi shows 2 times the signal: the blue one is the Output from the
Arduino, the yellow one is between resistor 1 and 2 (from here a direct
connection is available to P0.4 and P0.5.)
The yellow 0 Line is 1V upper the blue 0 Line. ( 0V is markeked on the left
side with 1 and 2)

4712, 2019-06-26:
Most likely the 20K pull down (maybe the capacitor too, it depends on the value) avoids the connection over the bootloader.

Wesie, 2019-06-26:
I don't think so, that the 20K pulldown has something to with that.
If the Signal 5V, so over the resistor(1) will be 0,49V and over resistor(2) will be 4,5V. In the Middle of both the MPU is connected and on P0.4 & P0.5 there are still 4.5V.  The Datasheet said, that the I/O Pins are 5V tolerant and High Signal will be recognise up from 2V. If I look on the picture from the Oszi, the flank at 2V (which are going direct to P0.4 &P0.5 (RX/TX) is a minimum longer (timebased) as the Input-Signal.
I'll try it once again with a 1K Pulldown (agains GND) Resistor on the Output of the Arduino.In this case the flank should be dropped faster.

Wesie, 2019-06-26:
I have just connected the small DSO203 (Mini Oszi) on the output of the Arduino:
Should the output Baudrate  to the ESC 74990 8N1 ?

4712, 2019-06-26:
19200 8n1.
Maybe the capacitor is even more important as shown [here ](https://www.rcgroups.com/forums/showpost.php?p=31759458&postcount=3889
)...

Wesie, 2019-06-26:
Thank's for the link, your time and help.
I'm not sure, if I should remove the capacitor or nor.... Its work like a filter during fly, if some noise on the signal-wire ..... After reading and see the picture from Oszi, I'm pretty sure that you're right, that the C is the reason for unexpected behavior.

I think, we can close the case with the solution, that the "C" must be removed.

closed 2019-06-26 by 4712